mercenaruss/zigstar_gateways #1: ESPHome Password

Julius112, 2021-08-31:
Hi Mercenaruss,

I really love your project and recently purchased one of your LAN Gateways. I've already added it to the Home Assistant Zigbee integration. However, to add it via the ESPHome integration I need to enter a password. What is the standard password of the pre-flashed Zigbee LAN Gateway? I couldn't find it anywhere in the docs.

Thanks a lot again for your great work.

Cheers,
Julius

mercenaruss, 2021-08-31:
Hi, thank you very much for support, sorry for this small misunderstanding, I did a .yaml template and is available for download, this way you can input your own password and setting. [YAML](https://github.com/mercenaruss/zigstar_gateways/blob/main/files/ESP-HOME/zigstar-gw-lan.yaml), after you generate firmware you do upload it by accessing LAN Gateway IP and doing OTA Update.

mercenaruss, 2021-08-31:
@Julius112 Firmware is update, you can download new .bin and upload using web interface and OTA.
Thank you for pointing this out.
[image]

closed 2021-08-31 by mercenaruss

Julius112, 2021-08-31:
Thanks a lot for your help!

Cheers,
Julius