mercenaruss/zigstar_gateways #2: [REQUEST] Zeroconf support to ZigStar LAN Gateway for Home Assistant ZHA network discovery

Hedda, 2021-09-23:
Please consider adding Zeroconf support to ZigStar LAN Gateway firmware for automatic ZHA network discovery.

Home Assistant OS (formerly HASSIO) support automatic network scanning and “Service Discovery” via Zeroconf and when a whitelisted Zigbee serial gateway is discovered it can pass along that to a domain like the ZHA integration component in Home Assistant.

For more details see:

https://community.home-assistant.io/t/zha-automatic-discovery-of-zigbee-coordinator-bridges-gateways-ethernet-wifi-network-devices-that-support-zeroconf-or-ssdp/293300

As a proof-of-concept this is now supported by Tube's Zigbee gateways (based on ESPHome firmware):

https://www.home-assistant.io/integrations/zha#discovery-via-usb-or-zeroconf

https://github.com/tube0013/tube_gateways

Once support for Zeroconf has been added to the firmware you also need the firmware to provide DNS TXT records for passing along setting parameters to integration config flows.

DNS TXT records could be used to pass along recommended settings parameters config flow to Home Assistant's ZHA integration:

radio_type=znp
baud_rate=value
data_flow_control=software
tcp_port_serial_gateway=value
Zeroconf DNS TXT records could also be used to pass along hostname, versions, location, MAC address, and more, etc..

http://www.zeroconf.org/Rendezvous/txtrecords.html

https://en.wikipedia.org/wiki/Zero-configuration_networking#DNS-based_service_discovery

https://en.wikipedia.org/wiki/TXT_record

More information about idea of using DNS TXT records for passing along setting parameters in https://github.com/thegroove/esphome-zbbridge#1 and https://github.com/thegroove/esphome-zha-ezsp-zeroconf

mercenaruss, 2021-10-11:
Work is done on this, but having small trouble submitting this to HA Repo.

Hedda, 2021-10-18:
@mercenaruss Sorry for the delay, I have now posted a reply to https://community.home-assistant.io/t/zha-automatic-network-discovery-of-zigbee-coordinator-bridges-gateways-ethernet-wifi-network-adapters-that-support-zeroconf/293300

If your Zeroconf and DNS TXT Records if correct then think you should only need to extend code in these two source files:

https://github.com/home-assistant/core/blob/dev/homeassistant/generated/zeroconf.py

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/manifest.json

The DNS server in your network router of course needs to support/enable mDNS (multicast DNS) so .local names can be used.

Hedda, 2022-02-03:
FYI, Zero-configuration networking (zeroconf) looks to have been changed and improved in the Home Assistant 2022.02 release:

https://github.com/home-assistant/core/pull/62133

https://www.home-assistant.io/integrations/zeroconf/

This change is only a concern for custom integration developers.

Currently zeroconf matching only allows matching the `macaddress`, `model`, and `manufacturer` properties along with the name from the ZeroconfServiceInfo.

Since properties are arbitrarily defined by the zeroconf service, the list of named properties has grown over time.

Matching now allows for any arbitrarily defined property. All property matches must be lowercase, wildcards are supported

The top level keys `model`, `manufacturer`, and `macaddress` are now deprecated from components `manifest.json` file and should be moved into a `properties` dict.

For example:

```
-    {"type":"_airplay._tcp.local.","model":"appletv*"}
+    {"type":"_airplay._tcp.local.","properties":{"model":"appletv*"}}
```

Hedda, 2022-02-06:
By the way, Home Assistant founders/leads now explicitly said this year they will prioritize focus on features like these types:

https://www.home-assistant.io/blog/2022/01/19/streamlining-experiences/

Check out this video at around 11 min 55 sec in where talk about that focus point about making it easier to get started, etc.

https://www.youtube.com/watch?v=t_2D_KoFIfU&t=710s

Hopefully, it should be a greater chance of getting more help from other Home Assistant developers with this if ask them.

Hedda, 2022-02-06:
FYI, PR https://github.com/home-assistant/core/pull/58224 for Home Assistant core has been closed since I cannot test it myself to confirm working or not.

It would be more than welcomed if anyone (who owns a ZigStar LAN Gateway) could take over to get zeroconf discovery in ZHA.

https://www.home-assistant.io/integrations/zha#discovery-via-usb-or-zeroconf

Again, other than Tube's Zigbee Gateways you can also checkout syssi's and thegroove's implementations as proof-of-concept:

https://github.com/syssi/esphome-zb-gw03

https://github.com/thegroove/esphome-zeroconf

https://github.com/thegroove/esphome-zha-ezsp-zeroconf

Hedda, 2022-03-30:
FYI, for reference, noticed that fairecasoimeme is working on a PR for Home Assistant which will add ZiGate discovery support:

https://github.com/home-assistant/core/pull/68577

"Add ZiGate device on automatic integration USB and ZEROCONF"

https://www.home-assistant.io/integrations/zha#discovery-via-usb-or-zeroconf

That will add automatic ZeroConf (MDNS) discovery of ZiGate Ethernet Gateway (not WiFi Gateway) + ZiGate USB adapters.

Hedda, 2022-05-10:
> FYI, for reference, noticed that fairecasoimeme from ZiGate is working on PR for Home Assistant to add ZiGate discovery support:
> 
> See -> [home-assistant/core#68577](https://github.com/home-assistant/core/pull/68577) ("**_Add ZiGate device on automatic integration USB and ZEROCONF_**")

FYI, PR for ZiGate's Zeroconf been merged in Home Assistant core making it an example for adding more a little more readable:

https://github.com/home-assistant/core/pull/68577

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/manifest.json#L85-L88

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/config_flow.py#L198-L201

https://github.com/home-assistant/core/blob/dev/tests/components/zha/test_config_flow.py#L80-L107

https://github.com/home-assistant/core/blob/dev/homeassistant/generated/zeroconf.py#L384-L389

mercenaruss, 2022-09-11:
@Hedda Zeroconf tested & PR submitted.

closed 2022-09-11 by mercenaruss