mercenaruss/zigstar_gateways #10: Зависания WEB

lavrizone, 2021-11-26:
Заметил зависания web части при активированной настройке MQTT, если остановить аддон Z2M, то вебморда шлюза очень плохо откликается на действия, страница обновляется раз в примерно 10 секунд, стоит Z2M запустить, то всё налаживается.

mercenaruss, 2021-12-01:
@xyzroe Anything we can do about this?

mercenaruss, 2022-01-16:
@xyzroe any idea about this?

xyzroe, 2022-01-16:
I have no idea what is common between the mqtt configuration and the state of the z2m addon.  

Web interface freezes are possible if there is no connection with the mqtt server. This is how the mqtt library works which is used in the firmware.

If someone has the opportunity to make issue in the correct [repository](https://github.com/xyzroe/ZigStarGW-FW/issues), this will help not to forget about it, at the time of working on the firmware.

closed 2022-01-17 by mercenaruss