mercenaruss/zigstar_gateways #12: Test and verify Koenkk experimental firmware for CC2652P configurable up to 20 dBm TX power output

Hedda, 2021-12-16:
FYI, Koenkk has now uploaded an experimental firmware for CC2652P/CC1352P with a default set to 9 dBm (and a max of 20 dBm):

https://github.com/Koenkk/Z-Stack-firmware/issues/323

He mentions in https://github.com/Koenkk/zigbee2mqtt/issues/8885 that he does not have RF measurement equipment so cannot confirm RF output power.

Would be great if you can test and verify Koenkk experimental firmware for CC2652P configurable up to 20 dBm TX power output.

Hedda, 2021-12-20:
FYI, Koenkk has now released mentioned +20 dBM RF power output enabled version as "Z-Stack_3.x.0 20211217" in the develop branch and closed issues as confirmed working with improved LQI as result:

[https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin)

[https://github.com/Koenkk/Z-Stack-firmware/issues/323](https://github.com/Koenkk/Z-Stack-firmware/issues/323)

[https://github.com/Koenkk/zigbee2mqtt/issues/8885](https://github.com/Koenkk/zigbee2mqtt/issues/8885)

That firmware has during the weekend been confirmed working by several community members with up to +20 dBM RF power output enabled, (tested in both Zigbee2MQTT and ZHA), and according to Koenkk it has also been confirmed working by ITead's own Sonoff Zigbee engineer using RF analyzer as well, so I think later if and when proven stable we can expect this version to ship with future batches.

Note that any releases in the develop branch should still be considered experimental until proven stable and any bugs specifically verified related to this firmware version should be reported as new issues to:

[https://github.com/Koenkk/Z-Stack-firmware/issues](https://github.com/Koenkk/Z-Stack-firmware/issues)

Char-r, 2021-12-23:
I tested on a Zigstar-LAN and I am not getting great results. none of my IKEA E1746 extenders show as online and most devices fail to respond to pings at all.

I went back to a SONOFF USB3 and the routers were found straight away so seems to a compatibility issue somewhere.

Hedda, 2021-12-23:
Are you testing [Z-Stack_3.x.0 20211217 firmware](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin) with Zigbee2MQTT or Home Assistant's ZHA? 

It does not appear to fully work yet with zigpy-znp which ZHA uses, see -> https://github.com/zigpy/zigpy-znp/issues/119

Hedda, 2022-01-12:
FYI, 20211217 has now also been moved to "master" branch and should be considered stable as recommended version:

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

By the way, the new "develop" branch got another also bug-fix:

https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/CHANGELOG.md

mercenaruss, 2022-01-16:
@Hedda hi,i'm back from vacation will start testing this firmware.
@Char-r Hi, please send me an email to see whats happening there. support@zig-star.com. Didnt get bad feedbacks,this will be one of the first,let see what is the problem.

mercenaruss, 2022-01-30:
Firmware was tested and is working properly.

closed 2022-01-30 by mercenaruss