mercenaruss/zigstar_gateways #14: Unable to disable WiFi

acovrig, 2022-03-17:
I have a ESP32-D0WDQ6 running 0.6.4 with TTGO T-Internet-POE. In settings I have the Disable emergency mode (API) checked and am connected to it over ethernet, but I still see the AP. Under status I also see that Ethernet is not connected, but that's how I'm seeing the page, so it is connected via Ethernet.

kolicheng, 2022-04-10:
I have the same problem.

mercenaruss, 2022-04-11:
@xyzroe you was explaining this once. I dont remember.

acovrig, 2022-04-11:
I was able to get it to show as connected and it then disabled the WiFi if I put it on a network with internet. If I put it on my IoT network w/out internet, then it assumes the local network is down and starts broadcasting WiFi.

closed 2022-04-11 by acovrig

xyzroe, 2022-04-15:
Pls go https://github.com/xyzroe/ZigStarGW-FW/issues/13#issuecomment-1099747614 to find all answers