mercenaruss/zigstar_gateways #15: 3.3V A short circuit     PCB problem C2 is incorrectly grounded in wiring and needs to be fixed. I also can't design the software, can't submit merge request

ahonxxx, 2022-04-18:
[image]

ahonxxx, 2022-04-18:
5V is also short-circuited
[image]

mercenaruss, 2022-04-20:
Hi,let me check and update today.
Thank you for pointing out this issues,nobody reported till now.

ahonxxx, 2022-04-22:
> 是否容易切断短路？我测试了还是不行，COM不能识别！

The copper sheet of the positive electrode of the capacitor needs to be cut off from top to bottom

65698888, 2022-04-22:
> 嗨，让我今天检查并更新。 感谢您指出这个问题，直到现在没有人报告。

> Hi,let me check and update today. Thank you for pointing out this issues,nobody reported till now.

Please inform  when updated, this is a very good project.

mercenaruss, 2022-05-08:
@ahonxxx @65698888 Problem solved,updated Gerbers are available.

closed 2022-05-08 by mercenaruss