mercenaruss/zigstar_gateways #20: Cannot connect to mqtt running as HA addon Client <unknown> disconnected, not authorised

evlo, 2022-09-07:
I'm trying to have zigstar GW connect to mosquitto MQTT server running as HA addon and getting these errors
[image]

```
2022-09-07 22:23:49: Client <unknown> disconnected, not authorised.
2022-09-07 22:24:34: New connection from on port 1883.
```

I tried these settings

[image]

and also
[image]

while having config like this
[image]

zigbee2mqtt connects and communicates no problem whatsoever

mmetzger1234, 2022-09-08:
Same issue, I  also noticed that the password field does not accept my entire long password on the mqtt config screen when i check the configMQTT.jason file. if i enter the full password in the config jason file . it seems to never want to try to connect.

evlo, 2022-09-08:
> Same issue, I also noticed that the password field does not accept my entire long password on the mqtt config screen when i check the configMQTT.jason file. if i enter the full password in the config jason file . it seems to never want to try to connect.

It happened to me one time too, but on second attempt it seemed to take full password.

mercenaruss, 2022-09-09:
@xyzroe any idea on this issue?

mercenaruss, 2022-09-09:
@evlo all software related issues,can be discussed here: https://github.com/xyzroe/ZigStarGW-FW

closed 2022-09-09 by mercenaruss