mercenaruss/zigstar_gateways #23: Broken links in repository

letovlive, 2022-10-13:
There are broken links on the website and its crosslinks after the last files rearrange.

letovlive, 2022-10-13:
[image]
[image]

letovlive, 2022-10-13:
the same 
[image]

I believe it is better to use some automated crawler to analyze broken links.

[image]

I've used integrity
https://apps.apple.com/us/app/integrity/id513610341?mt=12

mercenaruss, 2022-10-13:
Issues fixed,thank you for pointing out.
Now checking all broken links for all locations.

closed 2022-10-13 by mercenaruss