mercenaruss/zigstar_gateways #30: error handling request

nebbiadigiorno, 2023-06-03:
hi, i am tring to install a zigstar lan gateway in zha but without success.
the error in the log:

```
Logger: aiohttp.server
Source: components/zha/radio_manager.py:224
First occurred: 13:48:10 (1 occurrences)
Last logged: 13:48:10

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 1058, in request_callback_rsp
    return await callback_rsp
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 253, in start_network
    commissioning_rsp = await self.request_callback_rsp(
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 1055, in request_callback_rsp
    async with async_timeout.timeout(timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 85, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 235, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 146, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 180, in post
    return await super().post(request, flow_id)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 267, in async_configure
    result = await self._async_handle_step(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 367, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/usr/src/homeassistant/homeassistant/components/zha/config_flow.py", line 370, in async_step_form_initial_network
    return await self.async_step_form_new_network(user_input)
  File "/usr/src/homeassistant/homeassistant/components/zha/config_flow.py", line 376, in async_step_form_new_network
    await self._radio_mgr.async_form_network()
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 224, in async_form_network
    await app.form_network()
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 333, in form_network
    await self.form_network(fast=True)
  File "/usr/local/lib/python3.10/contextlib.py", line 217, in __aexit__
    await self.gen.athrow(typ, value, traceback)
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 875, in capture_responses
    yield queue
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 291, in start_network
    raise zigpy.exceptions.FormationFailure(
zigpy.exceptions.FormationFailure: Network formation refused: there is too much RF interference. Make sure your coordinator is on a USB 2.0 extension cable and away from any sources of interference, like USB 3.0 ports, SSDs, 2.4GHz routers, motherboards, etc.

```

i tried on the latest versione of home assisstant.
any ideas?

mercenaruss, 2023-07-06:
What setings you are using?

mercenaruss, 2024-05-07:
Closed due inactivity.

closed 2024-05-07 by mercenaruss