mercenaruss/zigstar_gateways #36: Zigbee destroyed?

Duffy987, 2023-09-19:
I tried to update my SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102(N) Zigbee USB stick.
Now it no longer works, it is no longer recognized by the Zigbee adapter or connected to the device or service is no longer active.

The following appears in the tool if you want to reinstall the firmware:
Timeout aiting for ACK/NACK after "Synch (0x55 0x55)"

Is there anything left to save?

Best regards

mercenaruss, 2023-09-19:
I need more detailed description of what options did you activate before flashing,to be able to tell you what happened.
Send here a screenshot of your settings for addon.

Of course,you can reflash it,device is fine if you didnt upload some custome firmware on him.

closed 2023-09-19 by mercenaruss

Duffy987, 2023-09-21:
Hello mercenaruss,

I tried my Zigbee stick:
SONOFF Zigbee 3.0 USB Dongle Plus Gateway, Universal Zigbee USB Gateway Base TI CC2652P + CP2102N
to flash.

I downloaded the firmware from Github.
I put the stick in boot mode and tried to flash it.

This error occurred like in the picture.

[image]

then I put the stick into boot mode again, plugged it in and tried again.
After that it worked.

But it always only works the second time.

Best regards, Duffy

mercenaruss, 2023-09-22:
Okay, so we talk about Multitool, not about zigstar addon.
Multitool still dont support automatical bootloader mode for Sonoff

Duffy987, 2023-09-22:
Many thanks for your response.
But can I assume that the stick is still installed cleanly if I try it a second time after the first attempt with an error message and then the success message appears?

Best regards, Duffy

mercenaruss, 2023-09-25:
Yeah,flashing is going well,except bootloader mode is not starting from first time.
We will solve this.

Duffy987, 2023-09-25:
Thank you for the Information.
Regards